OctoPrint/OctoPrint-FirmwareUpdater #363: Support for Prusa MK3.5

timur-tabi, 2024-08-11:
I have a Prusa MK3 upgrade to an MK3.5.  It appears that this plugin is incompatible with the MK3.5 (and later) control boards.  Is it possible to add support?

benlye, 2024-08-11:
I don't have the printer or the board so cannot say. 

- How is it updated normally?
- Can you put a firmware on it which exposes the SD card (assuming it has one) to a USB-connected host?

snitkofb, 2024-08-27:
The files changed format from .hex to .bbf  which doesn't appear compatible with the extension.

benlye, 2024-08-27:
The firmware updater plugin doesn't care about file format, other than filtering the list of files which can be selected. The filter can be disabled with this setting:

[image]

closed 2024-09-16 by github-actions[bot]