OctoPrint/OctoPrint-FirmwareUpdater #366: /media/usb - "The path is not writeable"

ubiquitous42, 2024-11-23:
**Hardware Setup**
Ender 3 Pro with SKR Mini v2.0

**Describe the problem**
I am able to successfully flash the firmware using an SD card inserted directly into the printer, but despite my best efforts to set up usbmount, I am unable to update the firmware via Firmware Updater.

**Log Files**
plugin-specific log file is empty
```
2024-11-22 18:38:20,173 - octoprint.server.api - ERROR - Error while testing if /media/usb0 is really writable
Traceback (most recent call last):
  File "/home/bocsweeney/oprint/lib/python3.9/site-packages/octoprint/server/api/__init__.py", line 549, in _test_path
    with open(test_path, "wb") as f:
PermissionError: [Errno 13] Permission denied: '/media/usb0/.testballoon.txt'
```

**Additional Info**
My user for octopi is not `pi`, but is instead `bocsweeney`.

jneilliii, 2024-11-23:
Did you setup the sudoers for that username? 

https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/doc/lpc176x.md#sudo-rights

```
bocsweeney ALL=NOPASSWD: /bin/umount
```

ubiquitous42, 2024-11-23:
@jneilliii I did, exactly as you typed it there.  No luck.

benlye, 2024-11-25:
You should try using the Marlin BFT option. It's much easier to configure.

ubiquitous42, 2024-11-27:
> You should try using the Marlin BFT option. It's much easier to configure.

In trying to switch to the Marlin BFT option, it insists that I don't have `marlin-binary-protocol` installed:
[image]

Despite following these instructions that were updated in Jun 2024:
[image]

Am I doing something obviously wrong?

benlye, 2024-11-27:
I expect you have not run the correct instance of `pip`. If you are using a venv (as you should be) then you must reference the pip command _inside_ the venv.

If you post your octoprint.log file I can probably figure out what correct path to pip is.

ubiquitous42, 2024-11-28:
That was indeed the issue, just the wrong version of pip.  Thank you for the quick response, I'm all set up now!

closed 2024-11-28 by ubiquitous42