OctoPrint/OctoPrint-FirmwareUpdater #367: Option for upload to a folder?

lsellens, 2025-03-07:
**Hardware Setup**
Ender 3 S1 using Marlin Binary File Transfer method

**Describe the problem**
The Ender 3 S1 requires the firmware be placed in a folder on the SD card named "STM32F4_UPDATE" Would it be possible to add an option to configure a folder in the upload path? It also needs a unique name so all it needs is the timestamp filename to be used in conjunction with the folder path.

**Additional context**
I did try using the dfu-utils, and stm32flash methods but I just get errors. If doesnt appear the bootloader supports dfu mode and I don't think the correct jumpers are set to use stm32flash.
dfu-utils returned 1
and with stm32flash
Cannot enter bootloader, check your BOOT0/Reset pins settings and try again

lsellens, 2025-03-22:
I'm very interested in this, so posting to keep the issue open.

jneilliii, 2025-03-22:
@benlye maybe flag as enhancement so stale bot doesn't kill this?

benlye, 2025-03-23:
Not sure if the binary protocol supports folders but I'll give it a go when I have a chance.

benlye, 2025-03-23:
This is not possible with the current BFT library, which throws an error if you try to pass anything other than a simple file name.

I don't know if the Marlin binary protocol itself would handle folders, and I don't really have a way to find out. Both the protocol and the library are more or less dead projects at this point, and taking them over is not something I'm going to do. It looks like we're stuck with what we've got.

closed 2025-03-23 by benlye

lsellens, 2025-03-24:
@benlye I appreciate you looking into it